ponty/pysimavr #6: Version 0.1.0 Seg faults in ubuntu (from pip)

chippography, 2015-09-23:
Installed pysimavr from pip (that's version 0.1.0).  I have the pre-reqs installed (Ubuntu 14.10 and tried 12.04, running in a Docker Container):

apt-get -y update
apt-get -y install python
apt-get -y install python-pip
apt-get -y install swig
apt-get -y install python-dev
apt-get -y install gcc
apt-get -y install libelf-dev
pip install pysimavr
# optional for examples:
pip install entrypoint2

When I run the test example:

```python
from pysimavr.avr import Avr
from entrypoint2 import entrypoint

@entrypoint
def run_sim():
    avr = Avr(mcu='atmega48', f_cpu=8000000)
    avr.step(1)
    print avr.pc
```

Causes a Segmentation Fault at the line ```avr = Avr(mcu='atmega48', f_cpu=8000000)```.

Running it again in gdb reveals:

```
root@302aaafbb9cd:~/tmp# gdb python
GNU gdb (Ubuntu/Linaro 7.4-2012.04-0ubuntu2.1) 7.4-2012.04
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
For bug reporting instructions, please see:
<http://bugs.launchpad.net/gdb-linaro/>...
Reading symbols from /usr/bin/python...(no debugging symbols found)...done.
(gdb) r test.py
Starting program: /usr/bin/python test.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff594e700 (LWP 10930)]
[New Thread 0x7ffff514d700 (LWP 10931)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff594e700 (LWP 10930)]
0x00007ffff6934131 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) where
#0  0x00007ffff6934131 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff5e845bd in SWIG_FromCharPtr (cptr=0xfffffffff5f19cc0 <Address 0xfffffffff5f19cc0 out of bounds>) at pysimavr/swig/simavr_wrap.c:3438
#2  _wrap_mem_logger_read_line (self=<optimized out>, args=<optimized out>) at pysimavr/swig/simavr_wrap.c:13429
#3  0x00000000004f95a5 in PyEval_EvalFrameEx ()
#4  0x00000000005008c2 in PyEval_EvalCodeEx ()
#5  0x0000000000501402 in ?? ()
#6  0x000000000045a8f6 in PyObject_Call ()
#7  0x00000000004fb364 in PyEval_EvalFrameEx ()
#8  0x00000000004f9d02 in PyEval_EvalFrameEx ()
#9  0x00000000004f9d02 in PyEval_EvalFrameEx ()
#10 0x00000000005008c2 in PyEval_EvalCodeEx ()
#11 0x00000000005012d0 in ?? ()
#12 0x000000000045a8f6 in PyObject_Call ()
#13 0x000000000043f52a in ?? ()
#14 0x000000000045a8f6 in PyObject_Call ()
#15 0x000000000045aba6 in PyEval_CallObjectWithKeywords ()
#16 0x00000000004bccc8 in ?? ()
#17 0x00007ffff7bc4e9a in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#18 0x00007ffff699f38d in clone () from /lib/x86_64-linux-gnu/libc.so.6
#19 0x0000000000000000 in ?? ()
```

nejsan, 2015-09-24:
I'm having the same issue, version 0.1.0 from pip on an Ubuntu 14.04 VM. At this point, I've narrowed it down to the call to `Uart` in `Avr`, but I haven't looked into where it's crashing inside the `Uart` class.

chippography, 2015-09-25:
Have you any luck compiling the ```HEAD``` of the repo?  I have not had any luck with it.  I  looked at some of the commits as to fixes with the segfault (see the commit history since 0.1.0), but it looks related to a the logger module.

nejsan, 2015-09-25:
I haven't tried compiling from `HEAD`. I managed to get 0.0.7 working on Ubuntu 12.04 but switched to using plain simavr because I didn't understand the pysimavr Arduino simulator.

I'll try to take a peek at some point because I think this project is incredibly useful, but for now I'll be working with simavr in C.

ponty, 2015-11-10:
I made a lot of updates, please try the latest version.
You can use Vagrant to get all dependencies in VirtualBox.

chippography, 2015-11-15:
Works with 0.2 on ubuntu 14.04

closed 2015-11-15 by chippography