ponty/pysimavr #9: Problems with additional libs

SebastianZug, 2016-03-22:
Hi,

I tested a very simple example for serial floating point outputs via serial communication. The most interesting feature is a printf functionality. A basic example is available on

[https://www.dropbox.com/s/06ua09zu4lsi9cq/AVR_writeSerial.c?dl=0](url)

For this purpose it is necessary to add 

`cc.options_extra = ['-Os', '-Wl,-u,vfprintf -lprintf_flt -lm']`

When I take the whole command and run it in a separate terminal, everything is fine. But pysimavr does not generate an appropriate elf file. This become obvious by the ? in the output. I made some tests and realized that this is caused by the variant of the subproces call in 212

`self.proc = Proc(cmd).call() `

If the subprocess parameters are contained in a string it works well

`subprocess.call('avr-gcc AVR_writeSerial.c -Df_cpu=16000000 -o /tmp/pyavrutils_wj1hbb.elf -mmcu=atmega2560 --std=gnu99 -Os -Wl,-u,vfprintf -lprintf_flt -lm', shell=True)`

but when we implement the other possibility and call the subprocess with a list 

`subprocess.call(['avr-gcc', 'AVR_writeSerial.c', '-Df_cpu=16000000', '-o', '/tmp/pyavrutils_wj1hbb.elf', '-mmcu=atmega2560', '--std=gnu99', '-Os', '-Wl,-u,vfprintf -lprintf_flt -lm'])
`
the test fails. What was the reason to call  self.proc = Proc(cmd).call() in line 212 with a list and how can this problem be handled?

Best wishes

SebastianZug, 2016-03-23:
The different results are not clear for me, but the problem is caused by the linker options. "-Wl" is not known to avr-gcc outside Atmel Studio. When I changed the subprocess call to

`subprocess.call(['avr-gcc', 'AVR_writeSerial.c', '-Df_cpu=16000000', '-o', '/tmp/pyavrutils_wj1hbb.elf', '-mmcu=atmega2560', '--std=gnu99', '-Os', '-uvfprintf', '-lprintf_flt', '-lm'])`

it generates floating point outputs :-)

closed 2016-03-23 by SebastianZug

ponty, 2016-03-25:
In the first case you use: `[..., '-Wl,-u,vfprintf -lprintf_flt -lm']`
This is not correct, all parameters should be separated like this: `[..., '-Wl,-u,vfprintf', '-lprintf_flt', '-lm']`
If you use the command line the shell makes the split, but there is no shell here.

In the scond case you use: `[..., '-uvfprintf', '-lprintf_flt', '-lm']`
This is correct.