ponty/pysimavr #11: Get the current Assembly Command

SebastianZug, 2016-04-15:
Hi ponty,

I want to run a step by step analysis of my elf file. Is it possible to access the assembly command for the current PC? Can I receive the SREG or any other register during execution?

Thanks for your help!

Sebastian

ponty, 2016-04-18:
**sreg** is an int array. I should check how swig handles arrays.

SebastianZug, 2016-04-18:
And what about the value of avr.pc? Its behaviour is strange, it looks more like a permanent incrementation than the address of the next opcode? I walked through the code but I cannot identify any problem :-(

ponty, 2016-04-20:
Please check the latest version for SREG.

```python
avr=Avr()
print avr.sreg
```

SebastianZug, 2016-04-22:
Yes, it works. Very nice. I started to run some additional examples an will deliver them soon.

Thanks a lot!

Sebastian

closed 2016-04-22 by SebastianZug