ponty/pysimavr #12: pysimavr on Ubuntu 16.04

SebastianZug, 2016-05-14:
Hi Ponty,

when I installed pysimavr on a 16.06 Ubuntu system and started an example I got 

`Traceback (most recent call last):
  File "hello.py", line 1, in <module>
    from pysimavr.sim import ArduinoSim
  File "/usr/local/lib/python2.7/dist-packages/pysimavr/sim.py", line 3, in <module>
    from pysimavr.avr import Avr
  File "/usr/local/lib/python2.7/dist-packages/pysimavr/avr.py", line 3, in <module>
    from pysimavr.logger import init_simavr_logger, terminate_simavr_logger
  File "/usr/local/lib/python2.7/dist-packages/pysimavr/logger.py", line 1, in <module>
    from pysimavr.swig.simavr import use_mem_logger, mem_logger_read_line
  File "/usr/local/lib/python2.7/dist-packages/pysimavr/swig/simavr.py", line 141, in <module>
    IRQ_FLAG_NOT = _simavr.IRQ_FLAG_NOT
AttributeError: 'module' object has no attribute 'IRQ_FLAG_NOT'`

I am not familiar with swig and I guess that something went wrong during installation. Any ideas?

Sebastian

ponty, 2016-05-26:
The old package had some swig2 generated files and you generated some files with swig3.
They are not compatible.
I fixed the setup process by removing swig usage completely.
All swig generated files are saved in the new package before uploading to pypi.
Please try the latest version!

ponty, 2017-02-24:
Fixed

closed 2017-02-24 by ponty