ponty/pysimavr #13: Attach function to interrupt

hharzer, 2016-08-18:
Hello,

Frost of all:  Great Project you're doing there!
I use it as a basis for an intern program to automatically test small arduino programs. 

In order to do that effectively, I want to attach a function to an interrupt via python. How would I do that? Adding a python function that is

gergoerdi, 2017-03-28:
I'd like to work on implementing this. However, I'll need some instructions on how to keep using the system-installed `simavr`, but change the `pysimavr` SWIG files and rebuild/reinstall the whole of `pysimavr`. @ponty can you help me get started?

Vincent14, 2017-05-12:
You should also ping the other contributors, perhaps they could be pleased to anwser you?

Premik, 2017-07-07:
This is bit old question. So just for completness... 
The question says "interrput" but I assume it was the simavr **IRQ** thing.  IRQ is a simavr name for an event distribution system and interrupts are part of this system.

It is possible to trigger events (raise IRQs) from Python and it is possible to hook a Python function/method to an IRQ. The later was only implemented recently and is not included in the latest release. Further if you need to mock out more than an IO port IRQ (digitalWrite hook) like ISP, ADC (analogRead), PWM (analogWrite) you also need the #33. Which includes few samples-ish here:
https://github.com/Premik/pysimavr/blob/a2cc8f620fc78dac7dcce0aca296699b505e89c4/tests/test_irq.py