ponty/pysimavr #15: Installation on Mac OS X El Capitain 10.11.6, Xcode 7.1.1

Vincent14, 2016-12-03:
Hi, I'm unable to install pysimavr, any idea?

```
# pip install pysimavr                
Collecting pysimavr
  Downloading pysimavr-0.2.1.tar.gz (1.1MB)
    100% |████████████████████████████████| 1.1MB 725kB/s
Collecting path.py (from pysimavr)
  Downloading path.py-9.0-py2.py3-none-any.whl
Collecting pyavrutils (from pysimavr)
  Downloading pyavrutils-0.2.tar.gz
Collecting easyprocess (from pyavrutils->pysimavr)
  Downloading EasyProcess-0.2.3.tar.gz
Collecting confduino (from pyavrutils->pysimavr)
  Downloading confduino-0.2.2.tar.gz
Collecting entrypoint2 (from confduino->pyavrutils->pysimavr)
  Downloading entrypoint2-0.0.6.tar.gz
Collecting munch (from confduino->pyavrutils->pysimavr)
  Downloading munch-2.0.4.tar.gz
Collecting pyunpack (from confduino->pyavrutils->pysimavr)
  Downloading pyunpack-0.1.1.tar.gz
Requirement already satisfied: six in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from confduino->pyavrutils->pysimavr)
Collecting decorator (from entrypoint2->confduino->pyavrutils->pysimavr)
  Downloading decorator-4.0.10-py2.py3-none-any.whl
Collecting argparse (from entrypoint2->confduino->pyavrutils->pysimavr)
  Downloading argparse-1.4.0-py2.py3-none-any.whl
Installing collected packages: path.py, easyprocess, decorator, argparse, entrypoint2, munch, pyunpack, confduino, pyavrutils, pysimavr
  Running setup.py install for easyprocess ... done
  Running setup.py install for entrypoint2 ... done
  Running setup.py install for munch ... done
  Running setup.py install for pyunpack ... done
  Running setup.py install for confduino ... done
  Running setup.py install for pyavrutils ... done
  Running setup.py install for pysimavr ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-build-KcXbuf/pysimavr/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-FTq444-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.macosx-10.11-intel-2.7
    creating build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/__init__.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/about.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/ac.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/avr.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/button.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/connect.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/firmware.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/inverter.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/lcd.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/ledrow.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/logger.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/proxy.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/sgm7.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/sim.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/spk.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/uart.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/udp.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/udpreader.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/vcdfile.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    creating build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/__init__.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/ac_input.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/button.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/hd44780.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/inverter.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/ledrow.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/sgm7.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/simavr.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/spk.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
[... 214 lines cut ...]
    copying pysimavr/swig/simavr/simavr/sim/sim_regbit.h -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr/sim
    copying pysimavr/swig/simavr/simavr/sim/sim_time.h -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr/sim
    copying pysimavr/swig/simavr/simavr/sim/sim_vcd_file.c -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr/sim
    copying pysimavr/swig/simavr/simavr/sim/sim_vcd_file.h -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr/sim
    creating build/lib.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr/sim/avr
    copying pysimavr/swig/simavr/simavr/sim/avr/avr_mcu_section.h -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr/sim/avr
    creating build/lib.macosx-10.11-intel-2.7/pysimavr/swig/simavr/tests
    copying pysimavr/swig/simavr/tests/tests.h -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig/simavr/tests
    running build_ext
    building 'pysimavr.swig._simavr' extension
    creating build/temp.macosx-10.11-intel-2.7
    creating build/temp.macosx-10.11-intel-2.7/pysimavr
    creating build/temp.macosx-10.11-intel-2.7/pysimavr/swig
    creating build/temp.macosx-10.11-intel-2.7/pysimavr/swig/simavr
    creating build/temp.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr
    creating build/temp.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr/sim
    creating build/temp.macosx-10.11-intel-2.7/pysimavr/swig/simavr/simavr/cores
    cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ipysimavr/swig/simavr/simavr/sim -Ipysimavr/swig/simavr/simavr -Ipysimavr/swig/include -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c pysimavr/swig/simavr_wrap.c -o build/temp.macosx-10.11-intel-2.7/pysimavr/swig/simavr_wrap.o --std=gnu99 -DNO_COLOR
    cc -fno-strict-aliasing -fno-common -dynamic -arch i386 -arch x86_64 -g -Os -pipe -fno-common -fno-strict-aliasing -fwrapv -DENABLE_DTRACE -DMACOSX -DNDEBUG -Wall -Wstrict-prototypes -Wshorten-64-to-32 -DNDEBUG -g -fwrapv -Os -Wall -Wstrict-prototypes -DENABLE_DTRACE -arch i386 -arch x86_64 -pipe -Ipysimavr/swig/simavr/simavr/sim -Ipysimavr/swig/simavr/simavr -Ipysimavr/swig/include -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -c pysimavr/swig/simavr_logger.cpp -o build/temp.macosx-10.11-intel-2.7/pysimavr/swig/simavr_logger.o --std=gnu99 -DNO_COLOR
    error: invalid argument '--std=gnu99' not allowed with 'C++/ObjC++'
    error: command 'cc' failed with exit status 1
   
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-build-KcXbuf/pysimavr/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-FTq444-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/tmp/pip-build-KcXbuf/pysimavr/
```

ponty, 2016-12-05:
I have no MacOS. I removed the gnu99 flag from setup.py.
Please check the latest version.

ponty, 2016-12-06:
to reproduce the problem on Ubuntu (trusty32):
```
CC=clang tox
```

ponty, 2016-12-06:
I think I fixed the problem, please test the latest version.

Vincent14, 2016-12-07:
I tried to pip install again but that's always the same version as 
previously, could you explain me how to install it without pip? I can't see 
any information in the Readme file.

Le mardi 6 décembre 2016 16:23:16 CET, ponty <notifications@github.com> a 
écrit :
> I think I fixed the problem, please test the latest version.
> 

-- 
Message envoyé via Dekko depuis mon appareil Ubuntu

ponty, 2016-12-08:
```
git clone https://github.com/ponty/pysimavr.git
cd pysimavr
pip install .
```

Vincent14, 2016-12-08:
I'm sorry to see a new problem:

```
$ pip install .
Processing /Users/vlambert/Documents/Personnel/cube/emulation/pysimavr
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-yr0K9d-build/setup.py", line 111, in <module>
        + files(CORES, 'sim_*.c', EXCLUDE),
      File "/private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-yr0K9d-build/setup.py", line 74, in files
        return [p for p in listdir(directory, pattern) if os.path.isfile(p) and os.path.basename(p) not in exclude]
      File "/private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-yr0K9d-build/setup.py", line 68, in listdir
        names = os.listdir(directory)
    OSError: [Errno 2] No such file or directory: 'pysimavr/swig/simavr/simavr/sim'
   
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-yr0K9d-build/
```

ponty, 2016-12-09:
Sorry, there is a git submodule also to clone.
```
git clone https://github.com/ponty/pysimavr.git
cd pysimavr
git submodule update --init --recursive
pip install .
```

Please also check the README for source build requirements (e.g. libelf-dev, python-dev, ..)

Vincent14, 2016-12-14:
I tried to brew install libelf but it was already here, and now I have this:

```
Requirement already satisfied: path.py in /Library/Python/2.7/site-packages (from pysimavr==0.2.1)
Requirement already satisfied: pyavrutils in /Library/Python/2.7/site-packages (from pysimavr==0.2.1)
Requirement already satisfied: easyprocess in /Library/Python/2.7/site-packages (from pyavrutils->pysimavr==0.2.1)
Requirement already satisfied: confduino in /Library/Python/2.7/site-packages (from pyavrutils->pysimavr==0.2.1)
Requirement already satisfied: entrypoint2 in /Library/Python/2.7/site-packages (from confduino->pyavrutils->pysimavr==0.2.1)
Requirement already satisfied: munch in /Library/Python/2.7/site-packages (from confduino->pyavrutils->pysimavr==0.2.1)
Requirement already satisfied: pyunpack in /Library/Python/2.7/site-packages (from confduino->pyavrutils->pysimavr==0.2.1)
Requirement already satisfied: six in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from confduino->pyavrutils->pysimavr==0.2.1)
Requirement already satisfied: decorator in /Library/Python/2.7/site-packages (from entrypoint2->confduino->pyavrutils->pysimavr==0.2.1)
Requirement already satisfied: argparse in /Library/Python/2.7/site-packages (from entrypoint2->confduino->pyavrutils->pysimavr==0.2.1)
Installing collected packages: pysimavr
  Running setup.py install for pysimavr ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-xBPiAh-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-Jv9mIA-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.macosx-10.11-intel-2.7
    creating build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/__init__.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/about.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/ac.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/avr.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/button.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/connect.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/firmware.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/inverter.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/lcd.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/ledrow.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/logger.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/proxy.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/sgm7.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/sim.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/spk.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/uart.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/udp.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/udpreader.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    copying pysimavr/vcdfile.py -> build/lib.macosx-10.11-intel-2.7/pysimavr
    creating build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/__init__.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/ac_input.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/button.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/hd44780.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/inverter.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/ledrow.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/sgm7.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/simavr.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/spk.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/uart_buff.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    copying pysimavr/swig/uart_udp.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/swig
    creating build/lib.macosx-10.11-intel-2.7/pysimavr/examples
    copying pysimavr/examples/__init__.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/examples
    copying pysimavr/examples/delay.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/examples
    copying pysimavr/examples/hello.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/examples
    copying pysimavr/examples/simple.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/examples
    copying pysimavr/examples/vcd.py -> build/lib.macosx-10.11-intel-2.7/pysimavr/examples
    running egg_info
    creating pysimavr.egg-info
    writing requirements to pysimavr.egg-info/requires.txt
    writing pysimavr.egg-info/PKG-INFO
    writing top-level names to pysimavr.egg-info/top_level.txt
    writing dependency_links to pysimavr.egg-info/dependency_links.txt
    writing manifest file 'pysimavr.egg-info/SOURCES.txt'
    warning: manifest_maker: standard file '-c' not found
    
    reading manifest file 'pysimavr.egg-info/SOURCES.txt'
[... cut ...]
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-xBPiAh-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-Jv9mIA-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-xBPiAh-build/

```

ponty, 2016-12-15:
You need **libelf** development version.
In Ubuntu it's called: **libelf-dev**

Vincent14, 2016-12-19:
Well, I can't find any version of this lib for Mac OS X on Google 

-- 
Message envoyé via Dekko depuis mon appareil Ubuntu

ponty, 2016-12-19:
Maybe you can install it from source: 
https://fedorahosted.org/elfutils/

ponty, 2017-02-24:
Fixed

closed 2017-02-24 by ponty