ponty/pysimavr #16: Installation on Mac OS X El Capitan 10.11.6, Xcode 7.1.1, Python 3.5

Vincent14, 2016-12-19:
I tried the same thing as in my previous ticket using another version of Python, is here the result:

```
$ pip3 install pysimavr
Collecting pysimavr
  Using cached pysimavr-0.2.1.tar.gz
Collecting path.py (from pysimavr)
  Downloading path.py-9.0.1-py2.py3-none-any.whl
Collecting pyavrutils (from pysimavr)
  Using cached pyavrutils-0.2.tar.gz
Collecting easyprocess (from pyavrutils->pysimavr)
  Using cached EasyProcess-0.2.3.tar.gz
Collecting confduino (from pyavrutils->pysimavr)
  Using cached confduino-0.2.2.tar.gz
Collecting entrypoint2 (from confduino->pyavrutils->pysimavr)
  Using cached entrypoint2-0.0.6.tar.gz
Collecting munch (from confduino->pyavrutils->pysimavr)
  Using cached munch-2.0.4.tar.gz
Collecting pyunpack (from confduino->pyavrutils->pysimavr)
  Downloading pyunpack-0.1.2.tar.gz
Collecting six (from confduino->pyavrutils->pysimavr)
  Downloading six-1.10.0-py2.py3-none-any.whl
Collecting decorator (from entrypoint2->confduino->pyavrutils->pysimavr)
  Using cached decorator-4.0.10-py2.py3-none-any.whl
Collecting argparse (from entrypoint2->confduino->pyavrutils->pysimavr)
  Using cached argparse-1.4.0-py2.py3-none-any.whl
Building wheels for collected packages: pysimavr, pyavrutils, easyprocess, confduino, entrypoint2, munch, pyunpack
  Running setup.py bdist_wheel for pysimavr ... error
  Complete output from command /usr/local/opt/python3/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-build-h54ab19a/pysimavr/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/tmpwn5ej4kvpip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.11-x86_64-3.5
  creating build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/__init__.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/about.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/ac.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/avr.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/button.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/connect.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/firmware.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/inverter.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/lcd.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/ledrow.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/logger.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/proxy.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/sgm7.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/sim.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/spk.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/uart.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/udp.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/udpreader.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  copying pysimavr/vcdfile.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr
  creating build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/__init__.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/ac_input.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/button.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/hd44780.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/inverter.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/ledrow.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/sgm7.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/simavr.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/spk.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/uart_buff.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  copying pysimavr/swig/uart_udp.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig
  creating build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples
  copying pysimavr/examples/__init__.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples
  copying pysimavr/examples/delay.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples
  copying pysimavr/examples/hello.py -> build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples
[... 519 lines cut ...]
    Skipping optional fixer: idioms
    Skipping optional fixer: set_literal
    Skipping optional fixer: ws_comma
    Fixing build/lib.macosx-10.11-x86_64-3.5/pysimavr/__init__.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/about.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/ac.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/avr.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/button.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/connect.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/firmware.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/inverter.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/lcd.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/ledrow.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/logger.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/proxy.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/sgm7.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/sim.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/spk.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/uart.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/udp.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/udpreader.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/vcdfile.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/__init__.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/ac_input.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/button.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/hd44780.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/inverter.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/ledrow.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/sgm7.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/simavr.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/spk.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/uart_buff.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/swig/uart_udp.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples/__init__.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples/delay.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples/hello.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples/simple.py build/lib.macosx-10.11-x86_64-3.5/pysimavr/examples/vcd.py
    Skipping optional fixer: buffer
    Skipping optional fixer: idioms
    Skipping optional fixer: set_literal
    Skipping optional fixer: ws_comma
    running build_ext
    building 'pysimavr.swig._simavr' extension
    creating build/temp.macosx-10.11-x86_64-3.5
    creating build/temp.macosx-10.11-x86_64-3.5/pysimavr
    creating build/temp.macosx-10.11-x86_64-3.5/pysimavr/swig
    creating build/temp.macosx-10.11-x86_64-3.5/pysimavr/swig/simavr
    creating build/temp.macosx-10.11-x86_64-3.5/pysimavr/swig/simavr/simavr
    creating build/temp.macosx-10.11-x86_64-3.5/pysimavr/swig/simavr/simavr/sim
    creating build/temp.macosx-10.11-x86_64-3.5/pysimavr/swig/simavr/simavr/cores
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Ipysimavr/swig/simavr/simavr/sim -Ipysimavr/swig/simavr/simavr -Ipysimavr/swig/include -I/usr/local/include -I/usr/local/opt/openssl/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python3/3.5.2_3/Frameworks/Python.framework/Versions/3.5/include/python3.5m -c pysimavr/swig/simavr_wrap.c -o build/temp.macosx-10.11-x86_64-3.5/pysimavr/swig/simavr_wrap.o --std=gnu99 -DNO_COLOR
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Ipysimavr/swig/simavr/simavr/sim -Ipysimavr/swig/simavr/simavr -Ipysimavr/swig/include -I/usr/local/include -I/usr/local/opt/openssl/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python3/3.5.2_3/Frameworks/Python.framework/Versions/3.5/include/python3.5m -c pysimavr/swig/simavr_logger.cpp -o build/temp.macosx-10.11-x86_64-3.5/pysimavr/swig/simavr_logger.o --std= [... cut ...]
Command "/usr/local/opt/python3/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-build-h54ab19a/pysimavr/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-e00czm4q-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/s4/2wvdkyc138n4t_zwk4wd59hr0000gp/T/pip-build-h54ab19a/pysimavr/
```

ponty, 2016-12-21:
You tested the same version of pysimavr, and you got the same error message as in #15.
Now I uploaded a bugfix version from Github to Pypi.
Please test it again.
You should get the libelf.h error, if you didn't install the lbelf headers.

ponty, 2017-02-24:
Fixed

closed 2017-02-24 by ponty