ponty/pysimavr #18: Loading hex files

skwee, 2016-12-23:
I understand that I can execute snippets with the Arduinosim. Is there a way to load a compiled hex file ?
or
I have a project with a handfull of c files. how can i simulate them with Arduinosim ?
Thanks Steve

Vincent14, 2017-05-12:
I'm also interested by this question, I allow myself a small ping at @ponty @Premik @SebastianZug, please =]

Premik, 2017-05-12:
There was some patch merged in the past. The ArduinSim now accepts **external_elf** as the last argument of the *__init__*. I think you can just  provide a .hex or .axf file name there.

Vincent14, 2017-05-12:
And what about a regular folder?

Premik, 2017-05-12:
I pressume a regular folder is an Arduino project. If you have multiple files I'm not sure how to do it. I'd personally just compile the sources using Arduino or avr-gcc separately and only simulate the compiled files. 

But you might check the  pyavrutils.arduino.**Arduino** class of the  [PyAvrUtils](https://github.com/ponty/pyavrutils) project directly. It is used by pysimavr internally. Perhaps it is able to compile multiple files.

Vincent14, 2017-05-12:
I searched how to generate a .hex file from the Arduino IDE and it's far more easy than I thought! I just followed this steps http://stackoverflow.com/a/11654333