ponty/pysimavr #19: Logger

Premik, 2017-02-15:
Hi,
When I enable debug level _logging.basicConfig(level=logging.DEBUG)_ the console got flooded with repeating message like: "atmega2560 reset"

Seems there is regression in the 47175c7ff68de93b570d8a97d012fcf20c52475c. 
The buf[256] got never truncated inside the _mem_logger_read_line_ so the last line is being read over and over.

ponty, 2017-02-22:
Can you show me your program?

Here is my program:
```python
import logging
from pysimavr.avr import Avr
from time import sleep

logging.basicConfig(level=logging.DEBUG)

if __name__ == "__main__":
    avr = Avr(mcu='atmega2560', f_cpu=8000000)
    sleep(2)
    avr.step(1)
    avr.step(1)
    sleep(2)
    avr.terminate()
```

output:
```
DEBUG:pysimavr.avr:mcu=atmega2560 f_cpu=8000000
DEBUG:pysimavr.logger:Starting atmega2560 - flashend 3ffff ramend 21ff e2end 0fff
DEBUG:pysimavr.logger:atmega2560 init
DEBUG:pysimavr.logger:atmega2560 reset
DEBUG:pysimavr.avr:terminating...
DEBUG:pysimavr.avr:...ok
```

ponty, 2017-02-22:
OK I found it. I had to rebuild the C extension.

Premik, 2017-02-22:
Yeah. I've actually already patched this one. Let me on PR.

ponty, 2017-02-24:
Fixed

closed 2017-02-24 by ponty