ponty/pysimavr #35: Add support for ATmega 2560

Vincent14, 2017-05-12:
I've seen a list of "arduino targets" [in this file](https://github.com/ponty/pysimavr/blob/master/pysimavr/avr.py#L24) where the AVR I need isn't. I've seen [a closed issue](https://github.com/ponty/pysimavr/issues/1) where @ponty added one directly in the repo.

Do you think it could be possible to do the same thing for [the ATmega 2560](http://www.atmel.com/Images/Atmel-2549-8-bit-AVR-Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf)? It has far more GPIO than others and my project is built on it.

...ping to the most active contributor @Premik with <3

Premik, 2017-05-12:
The atmega2560 is supported. Seems that ``arduino_targets`` field you are referencing is not really used for anyting now.
The close issue about ATmega8 was something else from what I can tell.

Vincent14, 2017-05-12:
This project looks awesome but the lack of documentation is a real pain =$ Thanks a lot for your precious help @Premik!

closed 2017-05-12 by Vincent14