ponty/pysimavr #37: Serial write after things get started?

gebl, 2017-05-21:
Hello,

I'm trying to work with getting python code to write strings over the Serial link over time.

I've set up a loop that looks like this:

while True:
    avr.run()
    r, w, x = select.select([sys.stdin], [], [], 0)
    if r:
      line = sys.stdin.readline()
      avr.uart.send_string(line+"\n")

This works for a short period of time, but, if i wait to type input or try again later, it is never delivered to the arduino. 

the arduino code looks like:

void loop() {
  Serial.print("Looping.\\n");
  Serial.print("Enter Value: \\n");
  while(1) {
    while (Serial.available() > 0) {
      char recieved = Serial.read();

      // Process message when new line character is recieved
      if (recieved == '\\n') {
          Serial.println(inData);
          inData = ""; // Clear recieved buffer
        } else {
          inData += recieved; 
       }
   }

Am i missing something? is there a better way to serial write to the arduino sim?

gabe

Vincent14, 2017-05-24:
Uh. Interesting. :( 

@ponty ? @Premik ? @SebastianZug ? Someone knows?

Premik, 2017-05-25:
Hi,
I can imagine you hit some simavr bug. Like this one: https://github.com/buserror/simavr/issues/144
It seems to me like once the **xoff** signal is generated the uart stops receiving bytes.
As a dirty workaround you can try to:

1.  Don't use the ``` while (Serial.available() > 0)``` call. Instead always just ```Serial.read()```. The return value would be -1 when there are no data.
2. In python send the data really **slooowly**. So the xoff never gets a chance to trigger. Replace the ```avr.uart.send_string(line+"\n")``` with:
```
for c in line+"\n":
            avr.uart.send_string(c)
            time.sleep(0.3)
```

- Note you don't neet do call the avr.run() in loop. It starts a background thread. Unless you paused it before.
- I suggest you to add one line at the end of the ```pysimavr.uart.send_string``` method. To not ingnore the return value (```x``` variable) from the backend. As it is used to indicate when the *uart_buf* device buffer is full.
```
def send_string(self, s):
        for c in s:
            x = write_fifo(self.backend.fifo_out, ord(c))
            if x < 0: raise BufferError("Avr io buffer full") #PATCH
```