ponty/pysimavr #38: No documentation on how to install from source/git

gergoerdi, 2017-07-25:
The `README.md` file shows the command `pip install pysimavr` for installation, but that command downloads a `pysimavr` release instead of installing the `git clone`'d source.

Please update the documentation with build/installation instructions when starting from a freshly cloned git repo.

gergoerdi, 2017-07-25:
I tried `pip install .` but then that fails to rebuild the SWIG files:

```
$ pip install .
Unpacking /home/cactus/prog/arduino/import/pysimavr/pysimavr.git
  Running setup.py (path:/tmp/pip-Sinnlq-build/setup.py) egg_info for package from file:///home/cactus/prog/arduino/import/pysimavr/pysimavr.git
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip-Sinnlq-build/setup.py", line 130, in <module>
        + files(CORES, 'sim_*.c', EXCLUDE),
      File "/tmp/pip-Sinnlq-build/setup.py", line 85, in files
        return [p for p in listdir(directory, pattern) if os.path.isfile(p) and os.path.basename(p) not in exclude]
      File "/tmp/pip-Sinnlq-build/setup.py", line 79, in listdir
        names = os.listdir(directory)
    OSError: [Errno 2] No such file or directory: 'pysimavr/swig/simavr/simavr/sim'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip-Sinnlq-build/setup.py", line 130, in <module>

    + files(CORES, 'sim_*.c', EXCLUDE),

  File "/tmp/pip-Sinnlq-build/setup.py", line 85, in files

    return [p for p in listdir(directory, pattern) if os.path.isfile(p) and os.path.basename(p) not in exclude]

  File "/tmp/pip-Sinnlq-build/setup.py", line 79, in listdir

    names = os.listdir(directory)

OSError: [Errno 2] No such file or directory: 'pysimavr/swig/simavr/simavr/sim'

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-Sinnlq-build
Storing debug log for failure in /home/cactus/.pip/pip.log
```

i do have SWIG installed:

```
$ swig -version

SWIG Version 2.0.11

Compiled with g++ [x86_64-unknown-linux-gnu]

Configured options: +pcre

Please see http://www.swig.org for reporting bugs and further information
```

Premik, 2017-07-26:
The *pysimavr/swig/simavr/simavr/sim* should contain the core simavr source. They are linked as a git submodule. Maybe they are missing from your source tree for some reason?

It is true there is very little dev documentation. Please also check:
- The **closed issues**. There has been various build related issues discussed and solved in the past.
- **Vagrantfile** in the root. Using Vagrant would get you a working environment easily.
- **.travis.yml** The whole build process is being done by Travis on each commit. So that would be another good place to check.

gergoerdi, 2017-07-26:
Yes indeed it was the missing `pysimavr/swig/simavr/simavr` submodule. Thanks!

I'm not closing this ticket yet because the documentation still needs updating.

gergoerdi, 2017-07-30:
Also, I've just now found out that SWIG-generated files like e.g. `pysimavr/swig/utils_wrap.cc` are checked in to git; moreover, if I remove them, `pip install .` doesn't seem to re-generate them.

What is the exact command line to re-generate the SWIG output? I'm asking because I'd like to upgrade the underlying `simavr` version, which of course contains API changes.

gergoerdi, 2017-07-30:
OK I figured out the way to do it is by running `scons` in `pysimavr/swig`.

joaoantoniocardoso, 2017-08-14:
Hi, I got some errors trying `scons` from `pysimavr/swig` dir:

`gcc -o simavr/simavr/sim/avr_usb.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_usb.c
gcc -o simavr/simavr/sim/avr_watchdog.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_watchdog.c
simavr/simavr/sim/sim_vcd_file.h:70: Error: Syntax error in input(1).
scons: *** [simavr_wrap.c] Error 1
scons: building terminated because of errors`

Premik, 2017-08-14:
Hi,
Seems the wrapper code generation finished ok it but failed to compile the generated code afterwards. It might be because of different swig version. But I'm not sure.
Could you try to reproduce this error in Vagrant ? There are some brief instructions on this comment:
https://github.com/Premik/pysimavr/pull/1#issuecomment-319592960

joaoantoniocardoso, 2017-08-15:
@Premik I didn't manage to run Vagrant with that Vagrantfile.

I'm using Arch Linux, I am able to install pysimavr using pip, but that isn't what I want...

`swig -version`:
`SWIG Version 3.0.12
Compiled with g++ [x86_64-pc-linux-gnu]
Configured options: +pcre`

`scons -version`:
`SCons by Steven Knight et al.:
	script: v2.5.1.rel_2.5.1:3735:9dc6cee5c168[MODIFIED], 2016/11/03 14:02:02, by bdbaddog on mongodog
	engine: v2.5.1.rel_2.5.1:3735:9dc6cee5c168[MODIFIED], 2016/11/03 14:02:02, by bdbaddog on mongodog
	engine path: ['/usr/lib/python2.7/site-packages/SCons']
Copyright (c) 2001 - 2016 The SCons Foundation`

Premik, 2017-08-15:
Vagrant woudl give you a working dev environment instantly. And could help troubleshooting any problems on your real box.  I'm an Arch user myself too and got no issues with running that Vagrantfile at all. What error are you getting?
```
> pacman -Q vagrant
vagrant 1.9.7-1
```
You actually don't need  to `pip install` the pysimavr. You can compile the extensions in place by `python2 ./setup.py build`. And use the `.o` from the `build` folder locally.
Or probably the virtual env. could help too.

On my system I'm using swig `3.0.13` so I believe the `3.0.12` works just fine. The compilation error you are getting is likely something else.

Are you trying to simply use pysimavr for your project or  are you trying to get dev environment so you can do improvements in the pysimavr itself?

joaoantoniocardoso, 2017-08-15:
Hi,

nice to hear that, here i'm using `vagrant 1.9.7-1` and `virtualbox-bin 5.1.22-1`.

`> vagrant up`
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Importing base box 'boxcutter/ubuntu1604'...
==> default: Matching MAC address for NAT networking...
==> default: Checking if box 'boxcutter/ubuntu1604' is up to date...
==> default: Setting the name of the VM: pysimavr_default_1502828612038_88643
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
==> default: Forwarding ports...
    default: 22 (guest) => 2222 (host) (adapter 1)
==> default: Booting VM...
There was an error while executing `VBoxManage`, a CLI used by Vagrant
for controlling VirtualBox. The command and stderr is shown below.

Command: ["startvm", "79110a8c-674e-4b0a-b2a1-377a8658f8e7", "--type", "headless"]

Stderr: VBoxManage: error: The virtual machine 'pysimavr_default_1502828612038_88643' has terminated unexpectedly during startup with exit code 1 (0x1)
VBoxManage: error: Details: code NS_ERROR_FAILURE (0x80004005), component MachineWrap, interface IMachine`

I'll try to update my virtualbox... and my system.

Premik, 2017-08-15:
Yeah, apparently a working vbox (or other vm backend) is a prerequisity to Vagrant...
I pressume you have the required kernel modules loaded. But just for the case:
```
sudo modprobe vboxdrv
sudo modprobe vboxnetflt
sudo modprobe vboxnetadp
```

joaoantoniocardoso, 2017-08-16:
Great tip, the virtualbox modules wasn't loaded. After a general aur update (and system reboot), the modules was successfully loaded and then finally it runs `vagrant up` with no errors. 

I'm totally new to vagrant. Now I should git clone and try there via ssh?

Premik, 2017-08-16:
Good stuff. 
No, it will automatically map your current source tree into the vm. So just ``cd /vagrant``. There are further details in the link I sent above.

joaoantoniocardoso, 2017-08-16:
Ok, I done the following:
`vagrant up`
`vagrant ssh`
`@vagrant > cd /vagrant/pysimavr/swig`
`@vagrant > apt install git pkg-config`
`@vagrant > rmdir simavr`
`@vagrant > git clone https://github.com/buserror/simavr`
`@vagrant > cd simavr`
`@vagrant > make`
`@vagrant > cd ..`
`@vagrant > cp simavr/simavr/sim_core_*.h include/`
`@vagrant > cp -r /usr/lib/avr/include/avr include/`
`@vagrant > scons`

And got the following:

> scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
swig -o simavr_wrap.c -Isimavr/simavr/sim -python simavr.i
gcc -o simavr/simavr/sim/avr_acomp.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_acomp.c
gcc -o simavr/simavr/sim/avr_adc.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_adc.c
gcc -o simavr/simavr/sim/avr_bitbang.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_bitbang.c
gcc -o simavr/simavr/sim/avr_eeprom.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_eeprom.c
gcc -o simavr/simavr/sim/avr_extint.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_extint.c
gcc -o simavr/simavr/sim/avr_flash.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_flash.c
gcc -o simavr/simavr/sim/avr_ioport.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_ioport.c
simavr/simavr/sim/sim_vcd_file.h:70: Error: Syntax error in input(1).
gcc -o simavr/simavr/sim/avr_lin.os -c -Wall -Wno-sign-compare --std=gnu99 -fPIC -DNO_COLOR -Isimavr/simavr/sim -Iinclude -Isimavr/simavr -I/usr/include/python2.7 simavr/simavr/sim/avr_lin.c
scons: *** [simavr_wrap.c] Error 1
scons: building terminated because of errors.

Premik, 2017-08-17:
So Vagrant  ruled any environment/library issues out. The only thing left is the source tree.
Note by doing git clone manually inside the simavr folder you are grabbing the very last simavr trunk instead of the fixed commit as this is supposed to work. 
If you want more recent simavr core [gergoerdi](https://github.com/gergoerdi/pysimavr/tree/update-simavr) did this already. All test are passing. I've merged it into my master branch.

I've tried doing everying fresh now:
```
git clone https://github.com/ponty/pysimavr.git
git submodule init
git submodule update
Cloning into '/tmp/work/pysimavr/pysimavr/swig/simavr'...
Submodule path 'pysimavr/swig/simavr': checked out '2062c2ddea13488d8d766331c5432b5bb2ae1fea'

vagrant up
vagrant ssh
```
Inside vm:
```
cd /vagrant/pysimavr/swig/
scons
```
Couple of warnings but it finished ``cons: done building targets.``.