ponty/pysimavr #39: Passing pin name to avr_alloc_irq

gergoerdi, 2017-07-27:
I'd like to expose the "pins" of my simulated off-MCU components to the MCU simulation as standalone `irq`s. In C(++), I can do that like this:

    namespace Name {
        const char* dc = "LCD.DC";
    }

    dc = avr_alloc_irq(&(avr->irq_pool), 0, 1, &Name::dc);

and then connect it to a pin of an MCU port:

    avr_connect_irq(avr_io_getirq(avr, AVR_IOCTL_IOPORT_GETIRQ('D'), 5), dc);

I can see that `avr_alloc_irq` is exposed in `pysimavr.swig.simavr`, however, I haven't been able to figure out how to pass the pin name to it. If I just try

    dc = avr_alloc_irq(avr.irq_pool, 0, 1, "LCD.DC")

then at runtime I get the following type error:

    TypeError: in method 'avr_alloc_irq', argument 4 of type 'char const **'

Of course, it's not hard to see why -- `"LCD.DC"` is a string, not a pointer to a string; but how do I get over this hurdle?

Premik, 2017-08-04:
Array of strings is probably something Swig can't handle out of the box. It would require some of the [typemap magic.](http://www.swig.org/Doc1.3/Python.html#Python_nn59)

Just wondering. Are you creating some new Part in C++ but then connecting it all in Python?
It is true all this IRQ connection facilities gives you some flexibility. Especially when you need to connect parts dynamically or chain them in runtime.
But for me it was always good enought to simply hook the callbacks to Python methods and raise IRQs from Python code. It is much easier.

gergoerdi, 2017-08-05:
My use case is creating new parts in Python, see e.g. https://github.com/gergoerdi/avr-chip8-simulator-py/blob/master/LCD.py 
The idea is that each part is implemented as a class that exposes some IRQs, and then there's a class for the board itself that connects them (see https://github.com/gergoerdi/avr-chip8-simulator-py/blob/d69ee2b0741ce60f561cee233c8e18e816fe500d/Board.py#L11 onwards) 

The benefit of adding names to the `avr_alloc_irq` calls would be so that they can be used with simavr's VCD facility with meaningful signal names.

gergoerdi, 2017-08-05:
Is that typemap stuff applicable to just `avr_alloc_irq`? If yes, I don't see a reason not to do that.

Premik, 2017-08-07:
I'm not sure about that typemaps. It is usually bit tricky business. 

I see. Allocating new IRQs and using the connect_irq is the official way to add new parts. If you need to connect some existing (C) parts together that is probably the way to go.

Regarding the nicer names. Perhaps renaming the AVR IRQ's where the part is connected to could achieve the same result? ``self.avr.irq.ioport_register_notify(self.onSce, ('D', 3)).get_irq().name = 'sce'``

I was trying to say it is not necessary to allocate new IRQs at all. I woudl probably make that Board class the only thing facing the pysimavr IRQs and doing the raise-irq calls and registering callbacks in there. And then leave all the other parts like LCD, Keyboard fully (py)simavr-agnostic. And just use some python level way to register/publish/distribute events.